tracespace/gerber-to-svg #22: Throws exception on non-gerber files

henrikekblad, 2015-09-25:
When running my test-files I got an exception when feeding the following files:
https://github.com/mgebundy/LilyPad-SR/tree/master/gerber

[Error: Error at line 17 - Cannot read property 'flashed' of undefined]
at module.exports (/Users/hek/git/nodebb-plugin-gerber3d/node_modules/gerber-to-svg/lib/gerber-to-svg.js:90:13)

mcous, 2016-05-02:
Just kidding this is still a problem

mcous, 2016-06-11:
As of 2.0.0 this particular file is not longer causing problems, so I'm going to close this issue again. This particular error was caused by an overly lenient regex in the drill file parser of gerber-parser. Will reopen or file new issues if this continues to be a problem.

closed 2016-06-11 by mcous

henrikekblad, 2016-06-11:
👍