tracespace/gerber-to-svg #25: Output coordinates factor 1000 smaller than gerber

TheunisKotze, 2016-03-14:
Running the attached gerber through the latest version of gerber-to-svg, I get the following svg header:
<svg width="1.0433in" height="0.748in" viewBox="0 0 1043.3 748">
(only the relevant bits)

The width/height attributes are correct, however the coordinates and thus viewbox seem to be a factor 1000 smaller than what would be expected. 
Is this intended? If so, any reason for the difference in magnitude? 
I've glanced through the code but have not yet found the cause.

This obviously doesn't affect rendering.

[outline.pho.zip](https://github.com/mcous/gerber-to-svg/files/172190/outline.pho.zip)

mcous, 2016-03-14:
This is intentional. The viewBox and all units inside the SVG are 1000 times the real-world units (first introduced 064ec7e). Firefox has issues with SVG coordinates that are very small or very large that result in SVG use tags (aka PCB pads) being rendered in the wrong position (see #13 and [this comment](https://github.com/mcous/gerber-to-svg/pull/10#issuecomment-68078776)). Multiplying the internal units and viewBox by 1000 works around the issue in all cases I could find.

1000 was picked for a few reasons. My first instinct was to just use the leading-zero suppressed integers as internal units, but that ran into a different Firefox bug where elements would disappear. It also meant two different files could theoretically have two different scales, which I didn't like. By using 1000, all Gerber units are now in mils (aka thous) or micrometers (depending on the original units), which is pretty easy for a human to read and understand, since they are commonly used manufacturing units.

TheunisKotze, 2016-03-14:
Sounds reasonable.  I'd suggest a note about that somewhere in the readme, perhaps here https://github.com/mcous/gerber-to-svg#things-to-watch-out-for

closed 2016-03-14 by TheunisKotze

mcous, 2016-03-14:
Yeah, I've been letting the current library / documentation languish as I've been doing a pretty serious refactor as I've had free time over the last year or so (currently available in the [release-candidate branch](https://github.com/mcous/gerber-to-svg/tree/release-candidate). It's noted in the new API documentation here: https://github.com/mcous/gerber-to-svg/blob/release-candidate/API.md#output