tracespace/gerber-to-svg #32: Fails to convert Gerbers generated by Orcad

alexgoldstone, 2016-07-10:
I have tried to convert Gerber files for a number of different PCBs using `gerber-to-svg` but all of my files fail. 

The most simple application (taken from your documentation) outputs the error message `cannot parse coordinate with format undefined`.

These Gerber files have all been sent to PCB manufacturers (and can be previewed in desktop Gerber viewer software) without problem.

I have attached a zip file containing a number of Gerber layers for one example PCB.

[KB0006R00.zip](https://github.com/mcous/gerber-to-svg/files/355990/KB0006R00.zip)

mcous, 2016-07-10:
Hey, I really appreciate the issue and files! The problem is being caused by the fact that these Gerber files have an invalid Format Specification line: `%FSLAN2X34Y34*%`. As per the [Gerber file spec](https://www.ucamco.com/files/downloads/file/81/the_gerber_file_format_specification.pdf) (see the table in section 6.1), the `N2` in that line is invalid. Because of those extra characters, the parser is missing the format details it needs. 

Obviously, though, seeing as these files exist and are manufacturable, the parser should be more lenient with this command. It should be an easy fix.

mcous, 2016-07-10:
I just released `gerber-parser@1.0.5` which fixes this issue, so you should be able to uninstall and then re-install gerber-to-svg to pick up the updated dependency

closed 2016-07-10 by mcous