tracespace/gerber-to-svg #34: Arcs from KiCad causing major rendering glitches

aziraphale, 2016-08-14:
Firstly I wanted to just say that I'm loving this project and have been using it to create a nice gallery page on my personal website showing all my PCB designs to friends :)

Unfortunately, when I started using KiCad's Arc tool to round off the corners of my boards (which itself is a monumentally-fiddly and unfriendly tool!), I found that arcs would completely destroy the output of `gerber-to-svg` in about 50% of cases (two of my boards have been fine; two have not). It's as if the SVG render takes the various lines and arcs that make up the board outline, and then joins them in completely the wrong order:

[image] [image]

Oddly it seems that uploading the same gerbers to OSH Park results in an "internal error", but uploading the KiCad `.kicad_pcb` file results in a correct rendering:

[image] [image]

Of course, given that OSH Park is also failing to render these gerbers, it's possible that KiCad itself is at fault and is producing invalid files. Alternatively, maybe `gerber-to-svg` and OSH Park both have the same bug (or OSH Park *is actually using* `gerber-to-svg` for its previews; I don't know).

I do know that **Gerbv** has no trouble rendering them correctly, so they can't be **too** broken...:

[image]

(KiCad's own **GerbView** and **Pcbnew**'s 3D viewer both display everything correctly, too)

Any ideas? I've attached the gerbers in question and my KiCad `.kicad_pcb` file, so hopefully this can be figured out :)

For reference, I've been using the latest KiCad nightly build, `2016-08-13 BZR 7029, Git cc2b358`.

Thanks

[gerbers.zip](https://github.com/mcous/gerber-to-svg/files/417112/gerbers.zip)
[dht22-adapter.kicad_pcb.zip](https://github.com/mcous/gerber-to-svg/files/417113/dht22-adapter.kicad_pcb.zip)

kasbah, 2016-08-14:
Hi, thanks for the very detailed report.

I put the Gerbers into a test on [tracespace/pcb-stackup](https://github.com/tracespace/pcb-stackup) (dht22-adapter-test branch) with a similar looking result. 

[image]

I had a similar experience recently with some KiCad produced Gerbers but with with a square outline where the edge points were only _slightly_ misaligned ([mattvenn/esp8266@931a026](https://github.com/mattvenn/esp8266-breakout/tree/931a026be2e97b9fe77931e6a4243a8eee7bda1e)).

mcous, 2016-08-15:
Hey @aziraphale, the reason you're seeing this is because (if you go into KiCad) itself, you'll notice the arc endpoints in the particular file don't precisely line up the endpoints of the line segments. So, when the SVG tries to fill in the outline loop, it gets the shape you're seeing instead of the completely filled in geometry.

We just pushed out pcb-stackup@2.0.0, which [exposes a new `outlineGapFill` option](https://github.com/tracespace/pcb-stackup/blob/master/API.md#options--object) to help with this. I was able to get your board to render with `{outlineGapFill: 0.005}`

mcous, 2016-08-17:
Closing this issue because I think you should be good with the outlineGapFill option (and it's more of a tracespace/pcb-stackup issue/question anyway). Let me know if that's not the case

closed 2016-08-17 by mcous

aziraphale, 2016-08-18:
Yes, sorry, that new option helped enormously, thanks :) Just not had the opportunity to respond until now!