tracespace/gerber-to-svg #35: CLI --color option doesn't actually do anything

mcous, 2016-12-30:
We set color here: https://github.com/mcous/gerber-to-svg/blob/master/bin/gerber2svg#L112

``` js
var getOpts = function(id) {
  var opts = {
    id: id,
    optimizePaths: argv['optimize-paths']
  }

  if (argv.color) {
    opts.color = argv.color
  }
  
  // ...
```

 I think `options.attributes.color` is what's actually supposed to be set

MCDELTAT, 2018-02-21:
I don't know if this project is still active but I would love to help fix this if I can just get some clarification on my issue. What I'm trying to do is load an SVG of just the pads of my design into Bantam Tools CNC Software. I'm exporting the gerbers from EAGLE, and then converting it to SVG in the CLI, but an interesting issue with the Bantam Tools software is that the geometry must be filled in order to be recognized, which has forced me to go into Inkscape and manually fill all geometries.

So my question is if there is a difference between color and fill. If there isn't, then I'm sure resolving this issue would help. Thanks. 

FYI when testing color I'm using this, which I'm not sure is the correct syntax:
```
gerber2svg -o out -c 00cc00 file.stc
```

mcous, 2018-02-21:
Hi, thanks for your interest! This project is less active than I'd like it to be, but unfortunately my free time isn't what it used to be.

Not knowing anything about Bantam Tools, I can tell you that the SVG's output by this tool have `stroke` and `fill` set to `currentColor` by default. On the web, `currentColor`'s default is black. Setting `color` on the root node (`<svg ... color="blue">`) changes the value of `currentColor`.

If you wanted to test whether fixing this particular ticket would help your use-case, take the outputted file, open it in a text editor, and add `color="black"` to the attributes of the root `<svg>`. Taking a stab in the dark, though, I suspect that #36 is probably more applicable for generating files that Bantam understands.

Do you have any documentation handy that describes how the software interprets SVGs?

MCDELTAT, 2018-02-21:
Yeah they have a page on their support site that talks about it but couldn't tell where the issue lies. That can be found here: https://support.bantamtools.com/hc/en-us/articles/115001668153-SVG-Files

Will try just manually setting the color and report back. I was attempting to understand what the fill tool in Inkscape does but unforunately it also adds a handful of other nodes such that running the following didn't help much.
```
diff inkscape_file.svg file.svg
```

MCDELTAT, 2018-02-21:
Update: After some experimenting with handwriting things and copying from the SVG created by Inkscape, the software I'm using definitely doesn't care about color. I also tried out some of the raw shapes but it doesn't seem to understand the x and y attributes because it stacked everything on top of one another. So in the end it seems the only thing it understands is Path tags. Here's an example of what Inkscape spits out after using the fill tool. So sorry for taking up your time. I'll probably continue doing it by hand for now and roll my own when I have time.
```
<path
     style="fill:#ff0000;stroke-width:0"
     d="M 1.0881846,5.9305189 C 0.04197981,4.8711547 0.04112694,4.869228 0.04112694,3.5648989 V 2.2595065 L 1.1013546,1.2124488 2.1615824,0.16539113 H 3.5673475 4.9731127 L 6.0201704,1.2256188 c 1.0462048,1.0593642 1.0470577,1.0612909 1.0470577,2.36562 V 4.8966312 L 6.0070005,5.9436889 4.9467727,6.9907466 H 3.5410076 2.1352423 Z"
     id="path194"
     inkscape:connector-curvature="0"
     transform="matrix(10.416666,0,0,10.416666,753.00008,113.00008)" />
```

mcous, 2018-02-23:
Thanks for the feedback! This is helpful for figuring out where to make improvements to the project. I think this means I should definitely implement #36 and have all the shape output be paths. If and when I find time to make those changes, I'll let you know

closed 2018-06-15 by mcous