tracespace/gerber-to-svg #40: SVG not being generated correctly for some gerbers

andersonls, 2018-01-24:
Hello, I'm having problems with some generated SVG using this tool. The generated SVGs are missing some elements with some gerbers. Opening the SVG with Inkscape i found that the elements seems to be being draw but with inverted colors.

Below you can find the Gerber used and the generated SVG (with gerber2svg) and PNG (with gerbv)

Gerber file: [gerber.txt](https://github.com/tracespace/gerber-to-svg/files/1660579/gerber.txt)

Excpected result (generated with gerbv):  [image]

Actual result (generated with gerber2svg): [generated-svg.txt](https://github.com/tracespace/gerber-to-svg/files/1660583/generated-svg.txt)

mcous, 2018-01-24:
Hi, sorry to hear you're having an issue! After spending a few minutes looking for differences between the gerbv and `gerber-to-svg` output, I was unable to find anything missing. Would you mind annotating the png below to point them out for me?

`gerber-to-svg` output run through [`svg2png`](https://github.com/domenic/svg2png):

[image]

andersonls, 2018-01-25:
That's weird, it seems like **inkscape** is not converting the SVG properly.

Do you know any other tool for converting SVG to high resolution bitmaps (2000 DPI and higher)? 

**svg2png** is taking almost 2 minutes to convert in 2000 DPI while **inkscape** takes 35 sec.

PNG generated with inkscape:
[image]

kasbah, 2018-01-25:
We've seen Inkscape compatibility issues before: https://github.com/tracespace/pcb-stackup/issues/29. If you can pin down the exact issue I'd be interested as I also use Inkscape to convert to PNG on [Kitspace](https://kitspace.org).

andersonls, 2018-01-25:
I highlighted the images bellow the elements that are missing when converting to PNG with Inkscape. If I open the SVG with Inkscape and ungroup the element you can see that they are present but for some reason Inkscape is drawing a black polygon on top of them.

Inskcape PNG missing elements: 
[image]

Inkscape interface:
[image]

Excpected result using gerbv:
[image]

mcous, 2018-01-25:
@andersonls the SVG output of `gerber-to-svg` appears to be correct. Like @kasbah said, this seems to be a problem with Inkscape not displaying the SVGs properly. I'm not an Inkscape user myself, but PRs to improve Inkscape compatibility are definitely welcome and helpful.

As for SVG to PNG conversion: `svg2png` uses phantomJS, which isn't my favorite thing in the world. You may have luck using [electron](https://electronjs.org/) or [Puppeteer](https://github.com/GoogleChrome/puppeteer), both of which could (theoretically) use Chrome to generate a PNG.

andersonls, 2018-01-26:
Ok, thank you. I just started messing up with RS-274X and SVG, but if I can find the issue i'll submit a PR.

closed 2018-01-26 by andersonls