tracespace/gerber-to-svg #43: Rotation and size

ju5t, 2018-04-30:
We have no control over the files we get or people sent to our website. This makes it challenging to design a box that holds the final svg on a website. Rotating svg's with css is no fun as it turns out. 

It would be of great help if we could force the longest side of the board to be either horizontal or vertical. Together with a maximum length (or width for that matter) it would make it much easier to include the svg in a website.

kasbah, 2018-04-30:
I have to re-iterate that I think people will be confused if you rotate their boards for them in a gerber viewer. You can fairly easily enforce a maximum width and height of your image container and scale the SVG accordingly.

mcous, 2018-04-30:
In addition to @kasbah's point about rotating design files being confusing to the user of a viewer tool, I'm thinking rotation of the image for display purposes is outside the scope of this library. I'm also pretty sure gerber-to-svg provides enough information (i.e. image width and height) for the library client to transform and display the image in any way it sees fit.

I might be able to be convinced otherwise, but at the moment I don't see a compelling reason to add a built-in image transformation feature

mcous, 2018-05-01:
@ju5t I think you'd have better luck doing the `transform` on the SVG and setting `width` and `height` on the parent: https://codepen.io/mcous/pen/odZrzV.

It does seem to me like the desired functionality is both easily accomplished in the library client and, like you said, a "personal" preference. Given that `gerber-to-svg` is, at its core, a 1-to-1 vector image to vector image converter, and I can't think of an elegant way to add "force the length or the width to be the longest length" to gerber-to-svg, I'm going to close this ticket. As always, I'll try to keep an open mind and I'd be happy to re-open / revisit if there really is need for this logic to live in the core conversion library.

Regarding the mirroring of the bottom image in `pcb-stackup`: the mirroring is an opinionated decision implemented by `pcb-stackup` outside of `gerber-to-svg`. Given that `pcb-stackup` is a `gerber-to-svg` client that wants to modify the images in certain ways, I feel that leaving transformation out of `gerber-to-svg` (and, for that matter, leaving additional transformation out of `pcb-stackup`) is internally consistent

closed 2018-05-01 by mcous