tracespace/gerber-to-svg #45: how to change color of svg

darkworks, 2018-05-23:
any idea how to change color of svg i tried with 

`--color "red" `

`--color "hex color code"`

`-- color "rgb color code "`
but none works

also how to pass width and height , tried 

```
--width "100%"  
--width 100%
```

poor documentation :(

mcous, 2018-05-23:
Sorry for the inconvenience! The CLI `--color` option is currently broken. It will be fixed when this project is moved to the `tracespace/tracespace` monorepo.

Closing this issue because it is a duplicate of #35

closed 2018-05-23 by mcous

mcous, 2018-06-15:
@darkworks `v4.0.0-next.6` of `gerber-to-svg` was just released (`npm install -g gerber-to-svg@next`). You may now specify color, width, height, and any other attribute you want to add:

```shell
gerber-to-svg --attr.color=blue --attr.width='100%' --attr.height='100%' -- some_file.gbr
```

darkworks, 2018-06-19:
can you paste v4.0.0-next github link

kasbah, 2018-06-19:
Everything has moved to a monorepo: https://github.com/tracespace/tracespace. Is that what you were asking for?

darkworks, 2018-06-19:
ya thanks